splintered-reality/py_trees_ros #16: ROS Wiki Docs Broken

stonier, 2017-03-13:
Half the [tutorials](http://docs.ros.org/latest/api/py_trees_ros/html/tutorials.html#tutorial-1-data-gathering) are just plain absent.

stonier, 2017-03-20:
Was missing a dependency on actionlib that was causing automodule docs to completely fail.

stonier, 2017-03-30:
On [Jenkins](http://build.ros.org/job/Kdoc__py_trees_ros__ubuntu_xenial_amd64/)

naveedhd, 2017-05-15:
the rosdoc_lite build has the rosdep init warning followed by bunch of autodoc import error:

```
# BEGIN SUBSECTION: rosdoc_lite - py_trees_ros
03:19:55 Invoking `rosdoc_lite` for package 'py_trees_ros': /tmp/rosdoc_lite/scripts/rosdoc_lite /tmp/catkin_workspace/src/py_trees_ros -o /tmp/generated_documentation/api_rosdoc/py_trees_ros -g /tmp/generated_documentation/symbols/py_trees_ros.tag -t /tmp/generated_documentation/rosdoc_tags/py_trees_ros.yaml
03:19:57 the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
```

Is this related?

stonier, 2017-05-15:
This particular problem has been driving me nuts.

That's a breadcrumb worth chasing though - py_trees' doc job doesn't have the same issue. I didn't catch that one - will dig.

stonier, 2017-06-24:
Finally working!

closed 2017-06-24 by stonier