splintered-reality/py_trees_ros #27: Error when using py-trees-blackboard-watcher

jr-bor, 2017-04-11:
I installed the py_trees suit from the ROS repository and when performing the [tutorial 3](https://stonier.github.io/py_trees_ros/tutorials.html#tutorial-3-blackboards) I get this error when executing any of the py-trees-blackboard-watcher related commands.

```
$ py-trees-blackboard-watcher --list-variables
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/py-trees-blackboard-watcher", line 23, in <module>
    py_trees_ros.programs.blackboard_watcher.main()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/py_trees_ros/programs/blackboard_watcher.py", line 222, in main
    handle_args(args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/py_trees_ros/programs/blackboard_watcher.py", line 167, in handle_args
    list_variables_service_name = find_service(args.namespace, 'py_trees_msgs/GetBlackboardVariables')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/py_trees_ros/programs/blackboard_watcher.py", line 142, in find_service
    service_name = rosservice.rosservice_find(service_type)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosservice/__init__.py", line 325, in rosservice_find
    t = get_service_type(s)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosservice/__init__.py", line 147, in get_service_type
    return get_service_headers(service_name, service_uri).get('type', None)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosservice/__init__.py", line 126, in get_service_headers
    raise ROSServiceIOException("Unable to communicate with service [%s], address [%s]"%(service_name, service_uri))
rosservice.ROSServiceIOException: Unable to communicate with service [/rqt_gui_py_node_26372/get_loggers], address [rosrpc://spaghetti-monster-machine:42717]
```

The spaguetti-monster-machine is a fictional machine name that tries to [boost performance](http://www.journals.uchicago.edu/doi/abs/10.1086/681096) by using the same sense of humor :smiley:

stonier, 2017-04-12:
I cannot reproduce here. Did fix it so that it actually reports the error (same exception gets thrown when there is no ros master, but different message) in eedeacdb.

That error though means the ros master is found and the system state is extracted, but as it's parsing over the list of services looking for a match, it bombs on that connection. What machine setup do you have? Is it over wireless?

jr-bor, 2017-04-12:
[Hmmm...](http://gph.is/2bQrBOY) now it works... 

Process: Turn on the computer -> clone repo -> test with launchfiles and script of your repo -> it works -> test with launchfiles and script of ROS repo -> it works -> try combination with both lasts -> it works -> I write this comment to close the issue.

It could be the couple of hundreds of overlapped workspaces that I have or the magic trick that fixes everything: turn off and on (º º_). Sorry for making you suffer :disappointed:

closed 2017-04-12 by jr-bor