splintered-reality/py_trees_ros #28: running_is_success -> ToBlackboard

naveedhd, 2017-05-11:
G'day.

Would it be a good idea to have a `subscriber` behaviour which returns `success` instead of `running` whether a message is received or not. I use `meta.running_is_success` to deal with non-frequent topics.

I checked the `subscribers.Event` does similar to intended case but it takes `std_msgs.Empty`. One way  of it would be to make it more general. 

What are your thoughts on this @stonier

stonier, 2017-05-15:
Do you mean [subscribers.EventToBlackboard](https://github.com/stonier/py_trees_ros/blob/devel/py_trees_ros/subscribers.py#L453) which always returns `SUCCESS`?

That one actually looks like a bug to be honest. It should return `SUCCESS` only when a msg is received so as to be consistent.

Key thing here is to provide as much information in the return value so that the user of the behaviour can decide what he wants to do with it. The combination of `RUNNING` when no msg is received and `SUCCESS` does that. The user can choose to infer, or simply to cap it with a decorator like you have done if they wish to filter the return information.

If you are proposing a new behaviour, or additional arguments to current behaviours to achieve it instead of the decorator, then I'd be worried that our decorators are failing to be useful enough to do the job they are designed to do. That feedback I'm interested in.

stonier, 2017-05-15:
A good design goal -> behaviours should be as few and as simple as possible with as much information as they can in the return value. Decorators are there to fill out the cartesian product of all desirable permutations arising from a sparse number of behaviours seeding that cartesian product.

naveedhd, 2017-05-15:
** cartesian product of all desirable permutations arising from a sparse number of behaviours seeding that cartesian product.** math!

closed 2017-05-15 by naveedhd