splintered-reality/py_trees_ros #31: Subscribers.ToBlackboard initialised with rosmsg()?

naveedhd, 2017-05-17:
Is there a way to feed the `initialise_variables` with the rosmsg instance?

for example:
```

pose_to_blackboard = ToBlackboard(topic_name="pose",
                                                 topic_type=geometry_msgs.msg.Pose,
                                                 blackboard_variables = {'pose': None},
                                                 initialise_variables=geometry_msgs.msg.Pose()
                                                 )
```

As I understand, we can supply a python dict which means initializing with dummy data would require rosmsg -> python dict conversion.

stonier, 2017-05-17:
I think what you're trying to do is:

```
pose_to_blackboard = ToBlackboard(topic_name="pose",
    topic_type=geometry_msgs.msg.Pose,
    blackboard_variables = {'pose': None},
    initialise_variables={'pose': geometry_msgs.msg.Pose()}
)

naveedhd, 2017-05-18:
exactly!

closed 2017-05-18 by naveedhd