splintered-reality/py_trees_ros #32: services.py?

naveedhd, 2017-05-17:
@stonier Would it be a good idea to have `services.py` similar to `actions.py`? could be good as a reference.

stonier, 2017-05-22:
Sounds ok. 

Make sure there are comments with flashing warnings though. Services should be used understanding the risks entailed. They are blocking, so there is potential to block your tree's tick-tock if you have a long running service, communication problem or even just enough service calls to exceed a threshold.

Use them, but sparingly or only if you need to. Better is the idea of a non-blocking service that acts like an action client with no feedback (python_rocon_comms). This is why we started the [service_pair](http://docs.ros.org/kinetic/api/rocon_python_comms/html/) pattern, but it doesn't have a c++ implementation yet so only works between python and python.

stonier, 2017-05-22:
Maybe this should go into the faq?

stonier, 2018-10-15:
Closing this, development moving on to ROS2 where there are asynchronous services (yay).

closed 2018-10-15 by stonier