splintered-reality/py_trees_ros #35: [actions.py] get_result().message

naveedhd, 2017-08-30:
Using the `ActionServer` behaviour for `move_base`, the tree will crash because of no `message` in `result` [here](https://github.com/stonier/py_trees_ros/blob/devel/py_trees_ros/actions.py#L99) with following:

```
File "/opt/ros/kinetic/lib/python2.7/dist-packages/py_trees_ros/actions.py", line 99, in update
    self.feedback_message = result.message
AttributeError: 'MoveBaseResult' object has no attribute 'message'
```
The `result` is empty and has no `message` field.

stonier, 2017-09-04:
Good catch. I wonder what action I was biasing that from? 

A useful solution might be to set a default handler (callback) for the setting of messages (e.g. "aborted", ...) and let the user specify a custom handler for it.

naveedhd, 2017-09-05:
It must be inspired from `py_trees_msgs/action`:  [Dock.action](https://github.com/stonier/py_trees_msgs/blob/devel/action/Dock.action) and [Rotate.action](https://github.com/stonier/py_trees_msgs/blob/devel/action/Rotate.action).

naveedhd, 2017-09-05:
However the `action_lib.set_succeeded/preempted/etc(result, msg)` pipes the `msg` to [actionlib_msgs/GoalStatus](http://docs.ros.org/api/actionlib_msgs/html/msg/GoalStatus.html) which is common and can be retreived by `action_client.get_goal_status_text()` in `rospy`. This simplifies things for us and the only custom message required by user (in most cases) would be feedback_message on running.

Following up this comment in PR https://github.com/stonier/py_trees_ros/pull/36

stonier, 2017-10-29:
Closed by #36.

closed 2017-10-29 by stonier