splintered-reality/py_trees_ros #40: Melodic release

veimox, 2018-08-21:
@naveedhd @stonier 

`py_trees` is released for melodic but `py_trees_ros`, `py_trees_msgs` and `rqt_py_trees` [are not](https://github.com/ros/rosdistro/blob/master/melodic/distribution.yaml#L2721).

I tried to create a workspace with these four and rosdep failed to find the `python-termcolor` package. This definition [was added](https://github.com/ros/rosdistro/pull/18809) and now it resolves correctly. Packages compile but I have not run any tests.

I was wondering what is the roadmap for these packages to be released in melodic. We could do a checklist and split resources :smile:

stonier, 2018-08-21:
* [ ] py_trees 0.6.1 ... on it's way
* [ ] py_trees_msgs 0.3.5 ... on it's way
* [ ] py_trees_ros
* [ ] rqt_py_trees

stonier, 2018-08-25:
@veimox can I lean on you for some melodic information (I don't have a pc with artful/bionic yet)? Trying to bloom this repo I get an error failing to find `python-termcolor`:

```
Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'python-termcolor' for distro 'bionic':
No definition of [python-termcolor] for OS version [bionic]
        rosdep key : python-termcolor
        OS name    : ubuntu
        OS version : bionic
        Data: debian:
  buster:
  - python-termcolor
  jessie:
  - python-termcolor
  stretch:
  - python-termcolor
  wheezy:
    pip:
      packages:
      - termcolor
fedora:
- python-termcolor
gentoo:
- dev-python/termcolor
osx:
  pip:
    packages:
    - termcolor
ubuntu:
  '*':
  - python-termcolor
  artful_python3:
  - python3-termcolor
  bionic_python3:
  - python3-termcolor
  cosmic_python3:
  - python3-termcolor
  trusty_python3:
  - python3-termcolor
  xenial_python3:
  - python3-termcolor
  zesty_python3:
  - python3-termcolor

Failed to resolve python-termcolor on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'python-termcolor', aborting.
python-termcolor is depended on by these packages: ['py_trees_ros']
<== Failed
```

That contains your updated keys, and melodic's [REP](http://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023) states that it should be targeting python2 still, but bionic's default is python 3 and I suspect that is what bloom is trying to configure. If that's the case, it's probably trying to look for a value that satisfies the `bionic_python3` key, but can't find `python-termcolor` there.

I notice that melodic is no longer being built for artful, so I think the right solution here is just to upgrade the rosdep keys to use `python3-termcolor`.

Why this worked for you is an interesting question. Are you on artful or bionic?

stonier, 2018-08-25:
Aha, I had an old rosdep that didn't support the wild card '*'.

stonier, 2018-08-25:
Everything on it's way so you should find them in the shadow repository this week.

Closing in advance so I don't have to return on the assumption that everything goes to plan :)

Will re-open if there are problems.

closed 2018-08-25 by stonier