splintered-reality/py_trees_ros #59: [melodic] py_trees_ros broken after last binary update

doisyg, 2019-02-13:
After updating last binary package in melodic (0.5.16-0bionic.20190204.215333), py_trees_ros seems broken. All my applications give me the same error:  `AttributeError: 'module' object has no attribute 'decorators'`

Even the first tutorial, here is the console output:

> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/py_trees_ros/tutorial_tree", line 44, in <module>
>     main_itself()
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees_ros/tutorials/one.py", line 139, in main
>     behaviour_tree.tick_tock(500)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees/trees.py", line 259, in tick_tock
>     self.tick(pre_tick_handler, post_tick_handler)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees/trees.py", line 233, in tick
>     node.visit(visitor)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees/behaviour.py", line 220, in visit
>     visitor.run(self)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees_ros/visitors.py", line 108, in run
>     self.tree.behaviours.append(conversions.behaviour_to_msg(behaviour))
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees_ros/conversions.py", line 122, in behaviour_to_msg
>     msg.type = behaviour_type_to_msg_constant(behaviour)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees_ros/conversions.py", line 49, in behaviour_type_to_msg_constant
>     elif isinstance(behaviour, py_trees.decorators.Decorator):
> AttributeError: 'module' object has no attribute 'decorators'
> [tree-10] process has died [pid 23311, exit code 1, cmd /opt/ros/melodic/lib/py_trees_ros/tutorial_tree one __name:=tree __log:=/home/**/.ros/log/6fd2029c-2f8c-11e9-949f-f0d5bfb298e1/tree-10.log].

stonier, 2019-02-14:
Aha, caught out. The experimental decorators module didn't get released on melodic. Version `0.6.7` is [on it's way](https://github.com/ros/rosdistro/pull/20307).  Until the deb is available, you can make use of the [py_trees/release-0.6.x](https://github.com/stonier/py_trees/tree/release/0.6.x) branch.

Thanks for the heads up & please re-open this if you continue to have problems.

closed 2019-02-14 by stonier

doisyg, 2019-02-14:
Thanks for your reactivity! Problem solved with [py_trees/release-0.6.x](https://github.com/stonier/py_trees/tree/release/0.6.x) compiled from source.