splintered-reality/py_trees_ros #60: ActionClient Behavior is not silent

airballking, 2019-02-19:
When running an Behavior of type ActionClient it always prints the goal that was send because of this line: 
https://github.com/stonier/py_trees_ros/blob/release/0.5-kinetic/py_trees_ros/actions.py#L92

To me, it would be desirable to disable this print out. I pollutes my console a bit.

If you agree, I could provide a pull request with a fix. I'd use a call such as ```self.logger.debug("{}".format(self.action_goal))```. Also, to which branch would you like to receive such a pull request?

stonier, 2019-02-20:
Aye, removed in #38 and released in melodic, but that didn't make it back to kinetic.

stonier, 2019-02-20:
Releasing in `v0.5.17` on kinetic.

closed 2019-02-20 by stonier

airballking, 2019-02-20:
Thanks a lot for the quick response and release!

PR #61 (2019-02-20): [actions] remove redundant print
This resolves #60.
commit 1e4c80c9 (2019-02-20): [actions] remove redundant print statement

This resolves #60.