splintered-reality/py_trees_ros #62: Resolve kinetic-melodic branches

stonier, 2019-02-20:
Really ought to have put them on the same `release/0.5.x` branch back when. They now seem to have gotten out of sync.

naveedhd, 2019-02-27:
Is there any signifant differences between melodic/kinetic? meaning can they be maintained as one?

I can start tracking kinetic (cause this is what we use) and start pulling it up to melodic (and later to crystal atleast in terms of features and idioms)

stonier, 2019-02-28:
It is perhaps possible. 

0.5->0.6 was dictated by a big push to make it python2/3 compatible (python2 got dropped later, 0.8 I think), but glancing at the changelog, it might not have any breaking api.

If you'd like to test 0.6 (melodic) sources on kinetic and they work out for you, we can get kinetic and melodic both on 0.6.

Backporting everything that's gone into crystal - that's significantly more work and much might not be possible if it breaks the api promise.

stonier, 2019-03-19:
Scratch above. I was staring at `py_trees` versions (kinetic: 0.5.x melodic: 0.6.x).

For `py_trees_ros` it should be fine .... They are both `0.5.x`. I'll move on this shortly.

stonier, 2019-03-23:
Both using the `release/0.5.x` branch now, new debs for both distros will be `v0.5.18`.

closed 2019-03-23 by stonier