splintered-reality/py_trees_ros #63: Timer not working inside a Selector after a failing action client behavior

doisyg, 2019-02-21:
Hi,
Still pretty new to py_trees, I am experimenting with ROS and I don't understand why Timers are not working after a failing action inside a selector. But are working after other type of failing behavior.
I have reproduced the problem with a modified version tutorial 5.

Below inside a selector with an always failing behavior, it is working as expected:
[image]

But with an always failing action client behavior, the timer is not working as expected:
[image]

Here is the modifed tutorial 5 code:
```
def create_root():
    # behaviours
    root = py_trees.composites.Parallel("Tutorial")
    topics2bb = py_trees.composites.Sequence("Topics2BB")
    scan2bb = py_trees_ros.subscribers.EventToBlackboard(
        name="Scan2BB",
        topic_name="/dashboard/scan",
        variable_name="event_scan_button"
    )
    battery2bb = py_trees_ros.battery.ToBlackboard(name="Battery2BB",
                                                   topic_name="/battery/state",
                                                   threshold=30.0
                                                   )
    priorities = py_trees.composites.Selector("Priorities")
    battery_check = py_trees.meta.success_is_failure(py_trees.composites.Selector)(name="Battery Emergency")
    is_battery_ok = py_trees.blackboard.CheckBlackboardVariable(
        name="Battery Ok?",
        variable_name='battery_low_warning',
        expected_value=False
    )
    flash_led_strip = py_trees_ros.tutorials.behaviours.FlashLedStrip(
        name="Flash Red",
        colour="red")

    scan = py_trees.composites.Sequence(name="Scan")
    is_scan_requested = py_trees.blackboard.CheckBlackboardVariable(
        name="Scan?",
        variable_name='event_scan_button',
        expected_value=True
    )
    scan_preempt = py_trees.composites.Selector(name="Preempt?")
    is_scan_requested_two = py_trees.meta.success_is_running(py_trees.blackboard.CheckBlackboardVariable)(
        name="Scan?",
        variable_name='event_scan_button',
        expected_value=True
    )
    scanning = py_trees.composites.Parallel(name="Scanning", policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
    test_timer = py_trees.composites.Selector("Test Timer")
    wait_5_secs =  py_trees.timers.Timer(name="Wait5secs", duration=5.0)
    wait_5_secs2 =  py_trees.timers.Timer(name="Wait5secs2", duration=5.0)

    #make the action client return failure for tests
    scan_rotate = py_trees.meta.success_is_failure(py_trees_ros.actions.ActionClient)(
        name="Rotate (always failing)",
        action_namespace="/rotate",
        action_spec=py_trees_msgs.RotateAction,
        action_goal=py_trees_msgs.RotateGoal(),
        override_feedback_message_on_running="rotating"
    )

    #failing behavior for alternative test
    failure = py_trees.behaviours.Failure(name="Failing behavior (not action)")

    scan_flash_blue = py_trees_ros.tutorials.behaviours.FlashLedStrip(name="Flash Blue", colour="blue")
    scan_celebrate = py_trees.composites.Parallel(name="Celebrate", policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
    scan_flash_green = py_trees_ros.tutorials.behaviours.FlashLedStrip(name="Flash Green", colour="green")
    scan_pause = py_trees.timers.Timer("Pause", duration=3.0)
    idle = py_trees.behaviours.Running(name="Idle")

    # tree
    root.add_children([topics2bb, priorities])
    topics2bb.add_children([scan2bb, battery2bb])
    priorities.add_children([battery_check, scan, idle])
    battery_check.add_children([is_battery_ok, flash_led_strip])
    scan.add_children([is_scan_requested, scan_preempt, scan_celebrate])
    scan_preempt.add_children([is_scan_requested_two, scanning])
    scanning.add_children([test_timer, scan_flash_blue])
    #With failing action, timer is not working
    test_timer.add_children([scan_rotate,wait_5_secs,wait_5_secs2])
    #Without action, working timer:
    #test_timer.add_children([failure,wait_5_secs,wait_5_secs2])
    scan_celebrate.add_children([scan_flash_green, scan_pause])
    return root
```

stonier, 2019-03-02:
Basic rules to pay attention to here:

**If a behaviour ticks with status `FAILURE||SUCCESS`, then it calls `terminate()` on itself and next time it's entered it calls `initialise()`**

For the action client, this means the tick immediately following it's failure, it re-initialises and starts trying to rotate again with status `RUNNING`.

**If a child under a selector that is previously `RUNNING` is interrupted by a higher priority child on the next tick, then the selector calls `terminate(INVALID)` on the lower priority child**

When the afore-mentioned action client re-initialises and returns `RUNNING`, it has interrupted the lower priority timer. The selector then calls `terminate(INVALID)` on the timer.

----

Compare the case with the always failing non-action client case. That higher priority `py_trees.behaviours.Failure` child *always* fails and subsequently, the interrupt does not occur (higher priority interrupts only occur when the higher priority child returns `RUNNING||SUCCESS`.

stonier, 2019-03-02:
Do you have a description of what logic you're trying to enable?

stonier, 2019-06-27:
Aging out. Reopen if you need to.

closed 2019-06-27 by stonier