splintered-reality/py_trees_ros #65: Kinetic apt package versions do not match

KKalem, 2019-03-04:
When ros-kinetic-py-trees-ros is installed through apt on Ubuntu 16.04 with ROS kinetic, the installed versions of py-trees and py-trees-ros do not match. py-trees is v0.5.10 and py-trees-ros is v0.5.17. This leads to decorators being unavailable in py-trees (They seem to have been added on 0.5.11?).

Everything just throws ` 'module' object has no attribute 'decorators' ` from `py_trees_ros/conversions.py` line 49. The line is: `elif isinstance(behaviour, py_trees.decorators.Decorator):`. 

I checked the install directory and indeed the file `decorators` does not exist.

stonier, 2019-03-04:
It's on it's way - http://repositories.ros.org/status_page/ros_kinetic_default.html?q=py_trees.

You'll notice that py-trees v0.5.12 is slated for the next public release. If you switch to using the `ros-shadow-fixed` repository, you'll be able to pick it up.

Re-open if you still have questions.

closed 2019-03-04 by stonier

stonier, 2019-03-04:
Looks like `0.5.12` got pushed to the public repository today :)

KKalem, 2019-03-05:
Confirmed. Everything works now :+1: 
Thanks a lot~