splintered-reality/py_trees_ros #107: Transform to blackboard behaviours

stonier, 2019-09-19:


stonier, 2019-10-21:
@peterpena do you have some example behaviour that alludes to what you're looking for? Failing that, basic inputs/outputs? 

If I had to hazard a guess to get started:

**Construction**

* two input strings on behaviour creation - the `from' and `to' links
* a variable name for storage on the blackboard

**Runtime**

* `SUCCESS` if a transform could be retrieved
* `RUNNING` or `FAILURE` depending on whether you would like the behaviour to be blocking or non-blocking if it could not retrieve the result (I would probably create both blocking and non-blocking counterparts for this)

peterpena, 2019-10-21:
I think this is good. Behavior will be something around the lines of: Is the position close enough to a frame of reference? Maybe adding a CheckTransform for bt will also be useful. So we can guard against any frames not existing and then we can get the transform safely with a success.

peterpena, 2019-10-21:
Being be able to add a transform from the behavior tree will also be good. Behaviors that will be useful for BT:

- AddTransform
- CheckTransform
- GetTransform

stonier, 2019-10-22:
**AddTransform** - #120

**CheckTransform & GetTransform**

You'll want to have the `Transform2Blackboard` behaviour early on in the tree tick and have other behaviours then just use the blackboard. This will make sure that you have a single valid transform value that doesn't change across the tick (and also simplify your other behaviours so that they don't need to handle any esoteric middleware api). In that sense, the `CheckTransform` and `GetTransform` can be just blackboard variable lookups with the usual blackboard behaviours.

**Other**

I think for the `Transform2Blackboard` behaviour, we should probably have a second variable which indicates whether the transform was extracted or not. Then the original transform variable can maintain the `last extracted transform`, which may actually have been extracted some ticks ago.

PR #121 (2019-11-07): [transforms] From/ToBlackboard behaviours added
Resolves #107 and #120
closed 2019-11-07 by stonier