splintered-reality/py_trees_ros #119: Best way to cancel action at shutdown?

pirobot, 2019-10-17:
(Originally [posted to py_trees](https://github.com/splintered-reality/py_trees/issues/242) by mistake instead of py_trees_ros.)

I need to cancel all move_base goals when I shut down my py_trees script. My move_base action behaviours use py_trees_ros.actions.ActionClient and I can move the robot around appropriately depending on the state of my behaviour tree. However, when I Ctrl-C my script, I need the robot to stop immediately so I need to cancel all move_base goals. I can do this by creating a standalone move_base action client via actionlib.SimpleActionClient in the main script and then calling cancel_all_goals() in my rospy.on_shutdown() function but this does not seem like a very "behaviour tree-like" way to do things.

Is there a better way using py_trees behaviors? I tried defining terminate() and/or stop() overrides for my move_base behaviors that then call move_base.cancel_all_goals() but these do not stop the robot from moving when doing a Ctrl-C on the main script.

naveedhd, 2019-10-29:
should work out of the box with `terminate()` using `py_trees_ros.actions.ActionClient`. which version of ros/py_trees_ros are you using?

stonier, 2019-10-29:
As @naveedhd mentioned, if you have a shutdown hook in the tree manager that calls `terminate(py_trees.common.Status.INVALID)` on the root behaviour, it should percolate to your running action clients and cancel them in their teriminate methods.

stonier, 2019-11-24:
Aging out. Reopen if you want to continue the discussion.

closed 2019-11-24 by stonier