splintered-reality/py_trees_ros #133: Question regarding Python 2 vs 3 and ROS 1 (Kinetic) vs ROS 2

pirobot, 2019-12-12:
We are very excited by all the active development being done here on py_trees, py_trees_ros and py_trees_js.  However, we need some advice on how to proceed given our current research environment here at the Stanford AI and Robotics lab.

Virtually all of our robots and desktops are still running ROS Kinetic on Ubuntu 16.04.  We do research on a wide variety of fixed arms (Sawyer, Franka, Kinova) and mobile manipulators such as the Toyota HSR, Turtlebot, Fetch and our own custom-built [Jackrabbot](http://svl.stanford.edu/projects/jackrabbot/).  However, it appears that most of the development on py_trees and py_trees_ros is being done for Python 3 and ROS 2 environments.  My question is what would you suggest is our best strategy for utilizing the latest features of py_trees, py_trees_ros and py_trees_js without doing a lab-wide upgrade of all our robots and computers to ROS 2?

Many thanks!

stonier, 2019-12-27:
Sorry for the delay, I really ought to check in on my emails more frequently!

**py_trees / py_trees_ros**

@naveedhd was working on slowly backporting some of the more interesting, newly added features of `py_trees` and `py_trees_ros` to the ROS1/python2 releases, but I think that's going to be a bit of work to iron out all the details (i.e. will be more than a few cherrypicks) as so much has moved forward.

I would be willing to help with reviewing PR's and answering questions (here, on chat or the occasional hangout) if you'd like to invest some effort into that though. I just wouldn't be able to support much rigorous testing of it though given a lack of ROS1/python2 environments in my workplace.

**py_trees_js / py_trees_ros_viewer**

Now a ROS1 version of the viewer using the same js library is probably quite doable. The js library has it's own 'json' specification for the data coming in and that is very compatible with the ROS1 trees. All you'd need to do is right a ROS1 version of `py_trees_ros_viewer` with it's own custom `ROS1` backend (c.f. the ROS2 [backend.py](https://github.com/splintered-reality/py_trees_ros_viewer/blob/devel/py_trees_ros_viewer/backend.py)) that converts messages into the format that the js library understands. If you're interested in spinning up a version of that, I can help there in making sure the js library has what you need.

pirobot, 2020-01-10:
OK, many thanks for the information.  Since we only have a passing need for behavior trees at this time, we will likely just code our own simple library to get by for now.

closed 2020-01-10 by pirobot