splintered-reality/py_trees_ros #150: In 0.5 there is a Qt dependency

boilerbots, 2020-02-20:
In your __init__.py you include **tutorials** which requires Qt. That is not good when I want this to run on a robot because Qt pulls in a other dependencies that I don't want. I see in later version of the project this was removed. Since **Melodic** is based on 0.5 can we get this fixed?

stonier, 2020-02-25:
Short of creating a separate package (which we did for later versions) I suspect the easiest workaround would be to do something like:

```
try:
    import python_qt_binding
    from . import tutorials
except ImportError:
    pass
```

That does still leave the problem of rosdep pulling in the qt dependencies though. I assume this is a problem for you also?

boilerbots, 2020-02-26:
I see Qt in the package.xml but it doesn't affect my project because I basically hand crafted the packages that are installed on the robot to keep it curated. 

I think for my short term solution I am going to install the package and then apply a **patch** over the project to remove the import lines from the project. As long as py_tree_ros doesn't change then this should work for now.

stonier, 2020-03-24:
Releasing the partial workaround (handles the import error, but doesn't handle the optional rosdep) in `v0.5.20`.

Keeping this issue open for visibility.

closed 2023-01-01 by stonier