splintered-reality/py_trees_ros #162: Release for ROS Noetic

UniBwTAS, 2020-06-29:
Will there be a release for ROS Noetic soon?

stonier, 2020-06-29:
In the middle of upgrading a laptop to `20.04` so I can play with it. It's going to need a few python3 patches, which shouldn't be too hard - I've run into those with the ros2 branches. So, TLDR; expect something soon. If you have PR's before I get to them, send them to the relevant `0.x.y` release branches. They will be the same as the ones I used for melodic - you can see them [here](https://github.com/splintered-reality/py_trees_ros/blob/devel/README.md).

I had thought of backporting the development that's gone into the `2.x.y` ros2 release branches, but that's moved forward so far now that I suspect it's going to be a major speed bump for anyone upgrading from melodic, and in most use cases, noetic will just be a maintenance upgrade for those users.

UniBwTAS, 2020-06-30:
Great to hear! Thank you, very much.

stonier, 2020-08-03:
Damn, that was a little bumpy, but finally ported. Had forgotten how much progress py_trees v2 has made.

Versions you're looking for (debs are rolling):

* py_trees `v0.7.0`
* py_trees_ros `v0.6.0`
* py_trees_msgs `v0.3.0`
* rqt_py_trees `v0.4.0`

That information is also in the matrix at the bottom of the README at https://github.com/splintered-reality/py_trees_ros along with links to the documentation.

closed 2020-08-03 by stonier

UniBwTAS, 2020-09-17:
Thank you for your great work! Maybe it is a good idea to make these packages available in the official ROS repositories. For Noetic these packages are not available yet. There is no tab for Noetic: http://wiki.ros.org/py_trees_ros

stonier, 2020-09-17:
The debs are available: http://repositories.ros.org/status_page/ros_noetic_default.html?q=py_trees

I suspect the wiki pages simply haven't had the script that updates it run over them yet.