splintered-reality/py_trees_ros #170: Installing on ROS1

pitosalas, 2020-11-25:
This is just too trivial but how do I install py_trees_ros on Melodic? I am getting weird attribute errors which lead me to conclude that it is not installed (correctly.) But sudo apt install ors-melodic- ... doesn't work in any variation. I am overlooking something but I am can't pin it down!

stonier, 2020-11-26:
I'm not actually using melodic, but did just enough to upgrade those branches for python3, pass tests and run demos so you might be running into some teething problems.

Got a script or example I could potentially reproduce your problems with?

stonier, 2020-11-26:
For reference, connected to #305

pitosalas, 2020-11-26:
Yes: 

#!/usr/bin/env python
import py_trees
bb = py_trees.blackboard.Client()

For example…Thanks…

Pito Salas
Faculty, Computer Science
Brandeis University

stonier, 2020-11-26:
Aye, just found your other issue :)

There's a large [compatibility matrix](https://github.com/splintered-reality/py_trees_ros) (bottom of the readme) that associates py_trees/platform/ROS version. If you're using the ROS build farm's deb for melodic, then it's 0.6.x for py_trees and 0.5.x for py_trees_ros. Looks like you're trying to use the blackboard api for `>1.x.y` versions, which would generate attribute errors.

Perhaps check you're trying to learn from the documentation for the correct version?

* http://docs.ros.org/en/melodic/api/py_trees/html/
* http://docs.ros.org/en/melodic/api/py_trees_ros/html/

pitosalas, 2020-11-26:
Ah I see! So the version that I get from ROS is the older version with the less 
fancy blackboard, yes?

Is there a way to get ROS to locate the newer version of py_trees?

Pito Salas
Faculty, Computer Science
Brandeis University

p.s. By the way, I forgot to compliment you on a really nice package, with good design, documentation, test cases, etc. Well done!

stonier, 2020-11-26:
Aye, re-inventing the blackboard to be something better than a bunch of unobservable global variables was the last big push to 1.0.

Will get back to you later today re adopting a newer version (family calling!)

pitosalas, 2020-11-27:
Hi Daniel

Thanks so much for your help!

Based on your feedback I tried to make a little progress by commenting out the blackboard stuff and using 0.6.8. 

There I found that setup_with_descendants doesn't exist yet (fine) and I changed it on a guess to `bt_root.setup(timeout=15)`. This produced (in Py 2.7) this error:

````
[CLOUD]~/catkin_ws$ rosrun prrexamples stroller_control.py 
('init ', 'Turn2Target')
('init ', 'ApproachTarget')
('init ', 'TrackFixedDistance')
[DEBUG] Find Wall            : Sequence.setup()
Traceback (most recent call last):
  File "/my_ros_data/catkin_ws/src/prrexamples/scripts/stroller/stroller_control.py", line 74, in <module>
    bt_root.setup(timeout=15)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/py_trees/composites.py", line 84, in setup
    new_result = child.setup(timeout)
TypeError: setup() takes exactly 1 argument (2 given)
[CLOUD]~/catkin_ws$
````

I mention this in case it is useful to you!

I look forward to your help in getting the 1.0 version to work in py2.7 (and if not, at least getting over this little hump :)

Best,

Pito Salas
Faculty, Computer Science
Brandeis University

stonier, 2020-11-27:
^Looks like you might have an ill-formed child. What type is that child that is failing? If it's your own class, then make sure it's inheriting and overwriting the setup method correctly (i.e. with timeout argument).

----

TLDR; It's likely there is no out-of-the box upgrade path for you on Melodic.

The 1.0+ libraries aren't compatible with the additional ROS1 tooling (messaging & services, blackboard & tree watchers, loggers, rqt_py_trees etc). They skipped straight to ROS2 and there is an entirely upgraded set of debug/viz tooling co-existing with the core libraries there.

That all could be backported, but there is a significant amount of effort required and ostensibly should be done by or in concert with someone making heavy use of them in ROS1 to iron out the wrinkles. I can't really do that without a local use case stress testing it, so opted not to do so to avoid making it unstable. There is also the problem of several companies relying on the pre-1.0 ROS1 api so a major version bump was out of the question. It could be released under a different namespace e.g. py_trees2 - but again, effort and instability.

On the other hand, if you're just planning to use `py_trees`, that's pretty portable, but would require a little bit of work. A few patches to make it python2 and catkin compatible.

pitosalas, 2020-11-27:
Hi Daniel

Thanks a lot for taking the time to look at it.

To play back my understanding -- py_tree_ros 1.0 is moved to ROS2 and going back is not an option

But Py_tree 1.0 has a dependency on python 3 which might not be too hard to reverse. Can you give me guidance on where the key changes would be and whether it would be worth attempting (that is cost of back porting vs. benefits of additional features etc in 1.0?)

Thanks!

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2020-11-28:
Hi Daniel

Is there a place where I can discuss use questions about py_trees. I don't think posting issues in the GitHub is the right way. Please advise? Thanks!

Pito Salas
Faculty, Computer Science
Brandeis University

stonier, 2020-11-30:
**Python 2/3 Compatibilty for PyTrees 2.x**

I don't think there would be too much involved in creating PR's for python2 compatibility on the `devel` branch. Tests and demos provide pretty good coverage, so you could start with them. The lads at CARLA are interested in this too. The critical piece is in making sure there is some CI job running tests to ensure it doesn't regress.

If you're just starting out though, I'd recommend sticking to the 0.6 version. Fundamentally it's the same api and workflow, with rough edges (especially around decorators & blackboard), so at least it'll get you familiar with api and development workflow before deciding you're seriously enough committed to working out what a backport looks like.

**Use Case Questions**

Hadn't really thought about making a chat room / forum available, but it could certainly use something - questions do trickle back here as issues and as you say, it's a rather sub-optimal way of fast-forwarding a conversation. I suspect there's even more that just don't get started. I think [gitter](https://gitter.im/) might serve as a lightweight experiment on that front. I'll see if I can set something up. In the meantime, you can ping me on my email (as listed in the `package.xml` in the root directory).

stonier, 2020-11-30:
If it wasn't for covid, we likely could have done :coffee: :)

stonier, 2020-11-30:
A gitter room now available at: https://gitter.im/splintered-reality/py_trees

If it wasn't for covid, we likely could have done :coffee: :)

pitosalas, 2020-11-30:
Thanks, see you on the list!

Pito Salas
Faculty, Computer Science
Brandeis University

pitosalas, 2020-11-30:
Hi Daniel

> Python 2/3 Compatibilty for PyTrees 2.x
>
> I don't think there would be too much involved in creating PR's for python2 compatibility on the devel branch. Tests and demos provide pretty good coverage, [...]

That's just what I did and I have something working. I will have detail questions about the exact behavior of control flow under different scenarios. Also I learned about catkin_virtualenv which would let me run py_trees. (Not py_trees_ros of course). 
> Use Case Questions
>
> Hadn't really thought about making a chat room / forum available, but it could certainly use something - questions do trickle back here as issues and as you say, it's a rather [...]

Thanks!

closed 2021-07-15 by pitosalas