splintered-reality/py_trees_ros #174: Add support for ROS2 Galactic

cco-seasony, 2021-07-07:
Hey!
Do you have any intentions on doing a release for ROS2 Galactic.

Thanks in advance!

stonier, 2021-07-12:
Aye, I do plan to. Moved house and bought another recently, so development hit a bit of a speed bump.

I'll probably upgrade to `py_trees 2.2` in the process. There's a couple of api changes needed.

Did you have try building from source? Any problems I should take note of?

cco-seasony, 2021-08-02:
@stonier Hey! sorry for the delay. We just moved from foxy to galactic and have not had the chance to build from source and test py tree.
We have some other things on the backlog before starting any testing with py tree. But I will let you know if we do so :D In the mean time I will keep track of your progress in case you upgrade before I build from source.

goekce, 2022-02-08:
I compiled the following repos from source on Galactic:
```
py_trees  py_trees_js  py_trees_ros  py_trees_ros_interfaces  py_trees_ros_tutorials  py_trees_ros_viewer
```

To test it I ran the tutorial `tutorial_two_battery_check_launch.py` but got an API-related error:

```
... configuration_group_box.py", line 58, in set_charging_status
[mock-dashboard-2]     self.ui.battery_charging_check_box.setChecked(charging_status)
[mock-dashboard-2] TypeError: setChecked(self, bool): argument 1 has unexpected type 'float'
```
Other nodes continued executing. I am new to `py_trees` so I am still not sure if `py_trees` can be used in Galactic.

jkeshav-bvignesh, 2022-05-24:
Hey @stonier, do you still plan to add ROS2 Galactic & Humble support? 
I assume that a colcon build from source won't fully work since you mentioned that a few api changes are required.

stonier, 2022-05-30:
Just started looking at it, ran into some speed bumps with rhel too. Probably going to need upgrade testing from nose to tox.

stonier, 2022-05-31:
Finally, underway. Tracking the effort for `py_trees` here: https://github.com/splintered-reality/py_trees/milestone/14.

Will do similarly for `py_trees_ros` when done there.

jodusan, 2022-08-16:
@stonier  hey, is galactic & humble support coming? Thanks!

stonier, 2023-01-01:
> I compiled the following repos from source on Galactic:
>
> ```
> py_trees  py_trees_js  py_trees_ros  py_trees_ros_interfaces  py_trees_ros_tutorials  py_trees_ros_viewer
> ```
>
> To test it I ran the [...]

Fixed in https://github.com/splintered-reality/py_trees_ros_tutorials/pull/47. There was a hidden bug present in the tutorial only.

stonier, 2023-01-01:
A meta-list for Humble

## Bugfixes

**py_trees_ros_tutorials**
* [x] https://github.com/splintered-reality/py_trees_ros_tutorials/pull/47

## New Features

## General Improvements

**py_trees_ros**

* #187 (eliminates warning messages)

**Development Environment**

See the README's in [py_trees](https://github.com/splintered-reality/py_trees/tree/devel/.devcontainer) and [py_trees_ros](https://github.com/splintered-reality/py_trees_ros/tree/devel/.devcontainer) for more info.

* [x] #183 (for foxy)
* [x] #187 (upgraded for humble)

stonier, 2023-02-04:
Humble / Rolling task list &rarr; #195.

closed 2023-02-04 by stonier