splintered-reality/py_trees_ros #177: Implementation on a custom bot on gazebo

tejalbarnwal, 2021-12-03:
I wish to use py_trees_ros with a custom bot on the gazebo. For the beginning, I thought of just implementing a very simple behaviour of some combination of going forward, backward, left or right.
How can I do this?
Has somebody earlier used py_trees_ros with gazebo apart from the ros navigation application?

stonier, 2021-12-03:
Aye, fairly common application. It's actually how we got started testing our fleet of robots. The py_tree application layer was agnostic whether it was a real robot or gazebo robot under the hood.

I don't have any ros2 code for this though. Basically you'll want to just set up the behaviours for communication with the ros2 navi interfaces (probably still actions?).

You can see a pretty similar example of this in:

* Code - https://github.com/splintered-reality/py_trees_ros_tutorials/blob/devel/py_trees_ros_tutorials/five_action_clients.py
* Docs - https://py-trees-ros-tutorials.readthedocs.io/en/release-2.1.x/tutorials.html#module-py_trees_ros_tutorials.five_action_clients

The scan rotate action for instance, is very much like a typical move base action call you'd put out to a gazebo robot.

tejalbarnwal, 2021-12-05:
Thank you so much for these resources
in case of doubts, I will continue on this thread.

stonier, 2022-05-30:
Closing out. Reopen if you wish to continue the discussion.

closed 2022-05-30 by stonier