splintered-reality/py_trees_ros #178: Trouble launching py_trees_ros tutorials

paulalcardoso, 2022-04-01:
Hi there! 

I'm trying to launch the tutorial one for py_trees_ros (in ROS Melodic) but I'm getting an error saying: " [tree-10] process has died [pid 16242, exit code 1, cmd /home/nta/catkin_ws/src/py_trees_ros/scripts/tutorial_tree one __name:=tree __". The log files says that there is an error importing module [one]. 

Do you have any suggestions on what I should try doing? I have this repository on another computer and all the tutorials are running fine, as well as the trees of my own.

stonier, 2022-05-30:
Sounds like it's not finding the correct python path. Are you sure you've got your environment configured correctly (would need steps to reproduce)?

stonier, 2023-01-01:
Aging out.

closed 2023-01-01 by stonier