splintered-reality/py_trees_ros #195: Humble Release Task List

stonier, 2023-02-04:
A meta-list for Humble

## Bugfixes

**py_trees**

* [x] https://github.com/splintered-reality/py_trees/pull/383
* [x] https://github.com/splintered-reality/py_trees/pull/366
* [x] https://github.com/splintered-reality/py_trees/pull/359

**py_trees_ros**

* [x] #191 
* [x] #194 

**py_trees_ros_tutorials**

* [x] https://github.com/splintered-reality/py_trees_ros_tutorials/pull/47
* [x] https://github.com/splintered-reality/py_trees_ros_tutorials/pull/48

## New Features / Breaking Changes

**py_trees**

* [x] https://github.com/splintered-reality/py_trees/issues/373
* [x] https://github.com/splintered-reality/py_trees/issues/352
  * `Sequence(memory=False)` (experimental&rarr;final)
  * `Selector(memory=True)` (experimental&rarr;final)
* [x] https://github.com/splintered-reality/py_trees/issues/364
  * `decorators.Repeat` (added)
  * `decorators.Retry` (added)
* [x] https://github.com/splintered-reality/py_trees/pull/376
  * `decorators.Count` (added)
  * `behaviours.Count` (removed &rightarrow; use `behaviours.StatusQueue` + `decorators.Count`)
* [x] https://github.com/splintered-reality/py_trees/issues/333
  * `idioms.eternal_guard` (removed &rightarrow; use `Sequence(memory=False)`)
* [x] https://github.com/splintered-reality/py_trees/issues/367
  * `behaviours.StatusSequence` &rightarrow; `behaviours.StatusQueue` (changed)

**py_trees_ros**

* [x] https://github.com/splintered-reality/py_trees_ros/pull/188
  * `tree.setup(node, node_name)` (added)

## General Improvements

**py_trees**

* [x] https://github.com/splintered-reality/py_trees/pull/360
* [x] https://github.com/splintered-reality/py_trees/pull/375

**py_trees_ros**

* [x] #187 (eliminates deprecation warnings)

## Development Environment

See the README's in [py_trees](https://github.com/splintered-reality/py_trees/tree/devel/.devcontainer) and [py_trees_ros](https://github.com/splintered-reality/py_trees_ros/tree/devel/.devcontainer) for more info.

* [x] #183 (for foxy)
* [x] py_trees - testing
  * https://github.com/splintered-reality/py_trees/pull/380
  * https://github.com/splintered-reality/py_trees/pull/386
  * https://github.com/splintered-reality/py_trees/pull/396
  * https://github.com/splintered-reality/py_trees/pull/397
* [x] py_trees - development environment
  * https://github.com/splintered-reality/py_trees/pull/389
  * https://github.com/splintered-reality/py_trees/pull/391
* [ ] py_trees_ros - development environment
  * [x] https://github.com/splintered-reality/py_trees_ros/pull/192
  * [ ] circleci -> github actions
  * [ ] update README
* [ ] py_trees_js - development environment
  * [ ] upgrade sphinx
  * [ ] circleci -> github actions
  * [ ] poetry
  * [ ] update README
* [ ] py_trees_js - development environment
  * [ ] upgrade sphinx
  * [ ] circleci -> github actions
  * [ ] deprecate use of distutils
  * [ ] check tutorials
  * [ ] update README

## Release

* [x] py_trees - [github](https://github.com/splintered-reality/py_trees/releases/tag/2.2.2), [pypi](https://pypi.org/project/py-trees/), [readthedocs](https://py-trees.readthedocs.io/en/release-2.2.x/),  [humble](https://repo.ros2.org/status_page/ros_humble_default.html?q=py_trees%24), [rolling](https://repo.ros2.org/status_page/ros_rolling_default.html?q=py_trees%24)
* [ ] py_trees_js - pypi, [humble](https://repo.ros2.org/status_page/ros_humble_default.html?q=py_trees_js), [rolling](https://repo.ros2.org/status_page/ros_rolling_default.html?q=py_trees_js)
* [x] py_trees_ros_interfaces - [humble](https://repo.ros2.org/status_page/ros_humble_default.html?q=py_trees_ros_interfaces), [rolling](https://repo.ros2.org/status_page/ros_rolling_default.html?q=py_trees_ros_interfaces)
* [ ] py_trees_ros - read-the-docs, humble, rolling
* [ ] py_trees_ros_viewer - humble, rolling
* [ ] py_trees_ros_tutorials - read-the-docs, humble, rolling
* [ ] ros webhook checks
* [ ] Discourse advertisement

boilerbots, 2023-02-17:
When will py-tree and py-tree-ros show up in rosdep for Humble?

stonier, 2023-02-22:
> When will py-tree and py-tree-ros show up in rosdep for Humble?

Completely depends on the ros team and when they roll out a release.

You can keep an eye on it [here](https://repo.ros2.org/status_page/ros_humble_default.html?q=py_trees). When the last box goes from red to green, it's been released.

You can also point your apt source to the staging apt server if you can't wait (or you just want to road test them). They're over at http://packages.ros.org/. It used to be called `ros-shadow-fixed`. Looks like you want `ros2-testing` now. You'll find the humble debs under there.

VArt3m, 2024-02-02:
@stonier 
Hello! Sorry to bother you, is there any plans to port or solutions for running `py_trees_ros_viewer` on ROS2 Humble? I've tried downloading the source and building it - which kinda works, but webview area remains empty and does not load.
I really would appreciate any pointer on what could be wrong there or what be done to port this great tool to Humble.
Thanks in advance

FranzEricSchneider, 2024-09-24:
Similarly, I'm trying to walk through the tutorials but the install steps are failing, my plan is to try to go backwards to Foxy for the tutorials, then maybe forward to Humble for usage unless the viewer seems critical.
```
E: Unable to locate package ros-humble-py-trees-ros-tutorials
E: Unable to locate package ros-humble-py-trees-ros-viewer
```

sea-bass, 2025-01-13:
The task list isn't quite complete as specified, but I'm getting going with releases to Humble, Jazzy, and Rolling.

Most of what remains is developer experience stuff anyhow, which can be done orthogonally to the actual binary releases to the newer ROS distros.