splintered-reality/py_trees_ros #210: I don't see py_trees_ros_interfaces service topic

boilerbots, 2023-02-28:
I could be totally doing something wrong but while waiting for the Humble release I added the latest source for py_trees_ros, py_trees, py_trees_js and py_trees_ros_tutorials to my projects src tree and built everything.

I can launch any of the tutorials and they all seem to run correctly but I can not run any of the tools such as py-trees-blackboard-watcher without getting the following error:

`py-trees-blackboard-watcher

ERROR: service not found [type: py_trees_ros_interfaces/srv/GetBlackboardVariables]
`

which makes sense because there isn't any py_trees_ros_interfaces service topic. For example, I launch

`
ros2 launch py_trees_ros_tutorials mock_robot_launch.py 
[INFO] [launch]: All log files can be found below /home/cmeyers/.ros/log/2023-02-28-14-37-34-187056-cmeyers-ubuntu-PF49RSRT-3207908
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: Bob the robot, at your service. Need a colander?
[INFO] [mock-battery-1]: process started with pid [3207909]
[INFO] [mock-dashboard-2]: process started with pid [3207911]
[INFO] [mock-docking-controller-3]: process started with pid [3207913]
[INFO] [mock-led-strip-4]: process started with pid [3207915]
[INFO] [mock-move-base-5]: process started with pid [3207917]
[INFO] [mock-rotation-controller-6]: process started with pid [3207919]
[INFO] [mock-safety-sensors-7]: process started with pid [3207921]
[mock-dashboard-2] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[mock-dashboard-2] [INFO] [1677623854.909222104] [dashboard]: 'battery' initialised
[mock-dashboard-2] [INFO] [1677623854.909552957] [dashboard]: 'get_safety_sensors' initialised
[mock-dashboard-2] [INFO] [1677623854.909726537] [dashboard]: 'set_safety_sensors' initialised
`

and I list the services

`
ros2 service list
/battery/describe_parameters
/battery/get_parameter_types
/battery/get_parameters
/battery/introspection/publishers
/battery/list_parameters
/battery/set_parameters
/battery/set_parameters_atomically
/dashboard/describe_parameters
/dashboard/get_parameter_types
/dashboard/get_parameters
/dashboard/introspection/publishers
/dashboard/introspection/subscribers
/dashboard/list_parameters
/dashboard/set_parameters
/dashboard/set_parameters_atomically
/docking_controller/describe_parameters
/docking_controller/get_parameter_types
/docking_controller/get_parameters
/docking_controller/list_parameters
/docking_controller/set_parameters
/docking_controller/set_parameters_atomically
/led_strip/describe_parameters
/led_strip/get_parameter_types
/led_strip/get_parameters
/led_strip/list_parameters
/led_strip/set_parameters
/led_strip/set_parameters_atomically
/move_base/describe_parameters
/move_base/get_parameter_types
/move_base/get_parameters
/move_base/list_parameters
/move_base/set_parameters
/move_base/set_parameters_atomically
/rotation_controller/describe_parameters
/rotation_controller/get_parameter_types
/rotation_controller/get_parameters
/rotation_controller/list_parameters
/rotation_controller/set_parameters
/rotation_controller/set_parameters_atomically
/safety_sensors/describe_parameters
/safety_sensors/get_parameter_types
/safety_sensors/get_parameters
/safety_sensors/list_parameters
/safety_sensors/set_parameters
/safety_sensors/set_parameters_atomically
`

Is something in Humble not starting a service in the expected namespace?

stonier, 2023-03-01:
> which makes sense because there isn't any py_trees_ros_interfaces service topic. 

It seems to be failing ot find the type definition of the service, not the service itself. 

`py_trees_ros_interfaces` is a package which provides the `GetBlackboardVariables` service definition. Do you have `ros-humble-py-trees-ros-interfaces` installed?

boilerbots, 2023-03-01:
I was using the released source package py_trees_ros_interfaces-2.1.0, but have updated to the latest branch and it still fails.
I have begun to write my own tree and I can launch it and both py-trees-blackboard-watcher and py-trees-tree-viewer work just fine.

My project is not launching by having the launch python import the source and use the generate_launch_description function as you have done in the demo scripts.

So I guess the problem could be isolated to the tutorials and not a general problem.

stonier, 2023-03-02:
Ah...digging a little further, you launched:

```
ros2 launch py_trees_ros_tutorials mock_robot_launch.py 
```

This is not launching a behaviour tree (and hence no service). It launches a few nodes to 'mock' a robot. The tree gets launched whenever you launch one of the  `tutorial_<number>_name.py` launchers. e.g.

```
ros2 launch py_trees_ros_tutorials tutorial_one_data_gathering_launch.py
```

boilerbots, 2023-03-02:
Ahh yes you are right, I failed at step 1, sorry about that. If I would have tried something else it would have worked.

closed 2023-03-02 by boilerbots