splintered-reality/py_trees_ros #216: Unable to locate ros-humble packages

olayasturias, 2023-08-22:
Hello,

When following the installation structures in your readme, I get the following errors:

```
E: Unable to locate package ros-humble-py-trees-ros-tutorials
E: Unable to locate package ros-humble-py-trees-ros-viewer
```

My OS is Ubuntu 22.04 jammy, and I am using ROS Humble distro.

Thank you!

AdoHaha, 2024-04-20:
Hi, still not working for humble

ldbrouwer, 2024-06-19:
Hello, any timeline for Humble tutorials and viewer completion?

slyandsmart, 2024-06-28:
I would also really appreciate a further development or some hits for humble.

FranzEricSchneider, 2024-09-24:
Still not available, also interested

pastoriomarco, 2025-01-02:
For those like me that need the viewer in ROS2 Humble, I could install it successfully following the instructions here:

https://github.com/splintered-reality/py_trees_ros_viewer/issues/36#issuecomment-2210393334

sea-bass, 2025-01-11:
I think the tutorials package is intended to be installed from source since you're supposed to tinker with the files locally, so no plans to release that.

As for `py_trees_ros_viewer`, I'm making the initial moves to get this on rosdistro: https://github.com/ros/rosdistro/pull/44153

If the package builds fine, we should be in good shape for next syncs of ROS 2 Humble, Jazzy, and Rolling.

sea-bass, 2025-01-13:
... actually, looking at the task list in https://github.com/splintered-reality/py_trees_ros/issues/195, it seems like the `py_trees_ros_tutorials` were also meant to be released? I guess I can get that going.

sea-bass, 2025-01-21:
Everything except `py_trees_ros_tutorials` is now available on ROS 2 Humble, Jazzy, and Rolling.

sea-bass, 2025-02-17:
... and `py_trees_ros_tutorials` is now available! Closing this.

closed 2025-02-17 by sea-bass