splintered-reality/py_trees_ros #224: New release to ros distro

fred-labs, 2024-08-23:
We plan to add a tool to ros distros that is based on py-trees/py-trees-ros: https://github.com/intellabs/scenario_execution

Unfortunately, we rely on the latest changes in `devel`-branch of py-trees-ros (especially #212) that are not yet part of a release.

Would you be able to create a new py-trees-ros release (e.g. 2.2.3), based on the latest `devel`?

sea-bass, 2025-01-13:
Yes! I'm in the process of releasing the new 2.3.0 -- all the PRs are up in `rosdistro` to Humble, Jazzy, and Rolling.

sea-bass, 2025-01-25:
This has been released in the latest Humble sync on January 21 -- will go ahead and (finally) close this.

closed 2025-01-25 by sea-bass

stonier, 2025-01-25:
Nice work @sea-bass! On behalf of everyone who's been asking, thanks!.